emoacht/SnowyImageCopy #12: Hi please help Exception error

markyder, 2015-01-31:
Hi I really want to use snowy to auto download pics from my camera but keep getting exception as per attached image, I have the flashcard set to station mode and am able to access via browser at http://flashair and see upload/ copy from no problems?

any help would be really appreciated, the machine is running win 7 with net 4.5.50709

thanks!

Mark
[image]
[image]

emoacht, 2015-02-01:
Hi, Mark,

Thanks for your report.
Let me look into it. Please wait for a while.

EMO

+++++
EMOTO Masayuki <emotomnf@nifty.com>

emoacht, 2015-02-01:
Hi Mark,

Thank you for your report.
Let me look into it. Please wait for a while.

EMO

emoacht, 2015-02-01:
Hi Mark,

I identified the code where the exception was thrown and it suggests that an unexpected file list data caused it. Something is wrong and I need to check the exact file list data. So, could you use /recordlog command line option and post the content of log file?

EMO

markyder, 2015-02-01:
Ok thanks no prob where does that log go?

Regards

Mark Deragon
RedDoor Homes

markyder, 2015-02-01:
?hi thanks for your help, below is th content of download.log

[cid:732f13ff-7842-4ac1-bb1c-f5f51fac6074]

markyder, 2015-02-01:
I've uploaded the file and exception just in case thanks !!
[image]
[image]

emoacht, 2015-02-02:
Thanks Mark,

I found the time stamp of "DCIM" folder is 2014-13-28. It is quite unusual and the cause of the exception. I can suppress the exception and am going to do so but wonder some strange things happening in your camera or FlashAir. So one more thing, could you check the update time or created time of "DCIM" folder from the Explorer?

EMO

markyder, 2015-02-02:
Hi strangely didn't have a date so deleted the folder and created another and all working !!1

Thank you so much!!

markyder, 2015-02-02:
Hi, the directory DCIM didn't have a date for creation so I deleted and recreated and all worked, Thanks
 is there a way to rename the card so you can use more than one on a network ? and one  other thing is there a way to download other file types ? ie docs etc?

Thanks again for your help !

emoacht, 2015-02-02:
Okay, anyway I will add some check algorithm to handle such a strange case.
Regarding FlashAir address, you can change address of each FlashAir card by editing CONFIG file in SD_WLAN folder. See [APPNAME](https://flashair-developers.com/en/documents/api/config/#APPNAME). Don't forget to set FlashAir address of Snowy to meet the address.
Regarding file types other than image files, I have no plan to add them at this moment because there will be so many things to think about to handle such files.

emoacht, 2015-02-12:
In ver 0.9.9, I added algorithm to handle such invalid date or time. If the item is a folder, it will not hinder the operation. If the item is a file, the file will be copied to a folder named "Unknown" in the specified local folder.

markyder, 2015-02-17:
Hi, am interested in doing some work with these wifi cards however would need some more functionality from the software for example software needs to see / copy all files

Is there a way to add multiple ssid's for the card to connect to.

I Would like to brand the software a bit

I Would like to add an auto format of the sd card option

Would like to call the devices web service something I could access over the Internet. And different for each one, maybe using no-ip as a DNS service.

Is this something you would be interested in performing for me, if so would you have a price in mind ? If all the functions could not be added which ones could?

Regards

Mark

markyder, 2015-02-20:
Hi did you receive my email as per below?

Hi, am interested in doing some work with these wifi cards however would need some more functionality from the software for example software needs to see / copy all files

Is there a way to add multiple ssid's for the card to connect to.

I Would like to brand the software a bit

I Would like to add an auto format of the sd card option

Would like to call the devices web service something I could access over the Internet. And different for each one, maybe using no-ip as a DNS service.

Is this something you would be interested in performing for me, if so would you have a price in mind ? If all the functions could not be added which ones could?

Regards

Mark

markyder, 2015-02-26:
Hi I just wondered if you received my email re possible doing some work (obviously on a paid basis) on your snowytool software

Regards

Mark Deragon
Director
RedDoor Homesn

closed 2015-12-25 by emoacht