emoacht/SnowyImageCopy #15: Question (Request)

OliverMattani, 2015-03-25:
Is it possible to make Snowy to copy only locked files?
On some cameras you can lock a file on a card.

emoacht, 2015-03-25:
Hi, Oliver,
Thanks for your question.
I am not sure what you mean by "lock a file on a card". My Canon camera has a function to "protect a image" in memory card. Are you talking about it? If so, it is technically possible to select such image files because they are ordinary files with ReadOnly attribute. It is what you want?
EMO

OliverMattani, 2015-03-25:
Hi EMO,
Yes I mean "protect a image" but it push a lock sign. sorry my mistake.
Is it possible that snowy just download protected files?
When I shoot I scroll the pictures and protect them on the fly, in mean time I would that snowy download them from my card to my laptop.

emoacht, 2015-03-26:
Okay, Added. You can set this app to select ReadOnly image only on check in new version to be released hopefully within a few days.

closed 2015-03-26 by OliverMattani