emoacht/SnowyImageCopy #31: Question about sorting the images

siburny, 2018-08-03:
Is there a way to do a reverse sort of the images? I would like the latest images to appear at the top.
Also, is it possible to just check for new images in the "Auto" (don't download them automatically)?

emoacht, 2018-08-05:
Hi, siburary,
> Is there a way to do a reverse sort of the images?
Currently no. But it sounds nice in some scenarios. I am working on it.
> Also, is it possible to just check for new images in the "Auto" (don't download them automatically)?
No. And I cannot imagine that auto without downloading images would be useful for most people. Usually, only checking images will not take a long time and there would be no substantial time saving by auto without downloading.

siburny, 2018-08-06:
> Currently no. But it sounds nice in some scenarios. I am working on it.
Thanks!

closed 2018-08-06 by siburny

emoacht, 2018-08-06:
Release Ver 1.4.0.
You can reverse the order of images by setting "Order from newer images" in "File".