emoacht/SnowyImageCopy #39: Hi~emoacht

dentphoto, 2020-02-11:
I am so happy because of your Snowy 1.7
There are two hopes.

1. In the Option "Create folder for each date"
   Current format is yyyymmdd.
   I want not only yyyymmdd but also yyyy-mm-dd.
   Is it possible?
2. In the Option "Sound"
   Another notification sound is also needed 
   when Snowy detects(connects) Flashair card??

Please help me.
Thanks in advance.

emoacht, 2020-02-21:
Hi, dentphoto
1. Regarding the formatting of date, it would be worth to consider.
2. Regarding the notification sound on detection or connection, actually this app just requests the card to respond and there is no clear point of such event in the work flow. It's not clear at all and this idea is not so fruitful in my view. So I decline it.

dentphoto, 2020-02-21:
Thanks for your kind reply.

When the Flashair is off, Snowy displays the following status messages sequentially.
      Waiting for auto check time
     -> Checking FlashAir
     -> Cannot connect to FlashAir

So I speculated that there would be a point when Snowy perform checking successfully 
when the Flashair is powered ON.
"Checking Completed" is the event of Flashair detection(connection).
Is it my misunderstanding??

[PS] Simple question
       What does "Timeout Duration" mean?
       I do not understand the exact meaning.
       Explain it to me please.

[PS] Example of wireless transfer with Your sound notification. ^.^
        http://www.youtube.com/watch?v=wZ3qrwIHXkM

emoacht, 2020-02-21:
Regarding the first question, it depends on how you define the detection or connection precisely.
For checking, there are several different sequences of steps depending on the situation and each step has its own time frame. 
See Check & Copy section of [Source/SnowyImageCopy/Models/Operator.cs](https://github.com/emoacht/SnowyImageCopy/blob/master/Source/SnowyImageCopy/Models/Operator.cs)

For the second question, the timeout duration acts as the maximum time length for each request from this app to the card. If this app cannot receive the response within this time length, the request will be automatically aborted. 
See FileManager.DownloadBytesAsync method of [Source/SnowyImageCopy/Models/ImageFile/FileManager.cs](https://github.com/emoacht/SnowyImageCopy/blob/master/Source/SnowyImageCopy/Models/ImageFile/FileManager.cs)

dentphoto, 2020-02-22:
Snowy 1.7 is a very powerful tool for my Flashair.
Thank you very much for your help.

dentphoto, 2020-02-23:
Sorry but last question.^^

In my opinion, the moment the message "Cannot connect to FlashAir" disappears 
is the very moment of "Flashair Connected".
Is it right?
If so, is it possible to define that moment?

emoacht, 2020-03-02:
It is wrong. The message will be shown when the connection fails any time during the connection.
I suggest you to go back to the original purpose of your suggestion and think about the straight forward way to accomplish it.

closed 2020-03-02 by emoacht