emoacht/SnowyImageCopy #41: I have another request on Snowy.

dentphoto, 2020-04-11:
Hi, Emoacht !

I have two flashair cards(Sta mode) on two digital cameras.
Two Snowy app windows are opened on the PC and each card successfully transmit images wirelessly .
Like the attached photo...
Is it possible that the Snowy app supports multiple cards in an integrated way in the future?
Without opening two or more Snowy app windows...

[image]

emoacht, 2020-07-16:
Thanks for your suggestion.
The current version is not designed to run multiple instances simultaneously. You can start multiple instances simultaneously, it is simply because doing so is not prevented internally, but the settings file and other files which this app creates will be shared within multiple instances. It may cause some unexpected behaviors.
At the same time, I understand the need to access multiple FlashAir cards simultaneously. I am currently working on the next version targeted for Microsoft Store which can open multiple windows for each FlashAir card. I believe that a window for a card is better than a window for multiple cards because it will be too complex and hard to manage.

dentphoto, 2020-08-01:
Thank you.
I look forward to the multiple card support version.

emoacht, 2020-08-12:
The new version becomes available at Microsoft Store.

closed 2020-08-12 by emoacht