emoacht/SnowyImageCopy #47: Is it a bug in multiple cards version?

dentphoto, 2020-08-18:
I am very satisfied with the flashair multiple cards version.
Pictures from each of the two cards are transferred well.
But there seems to be a problem.

When I close the FLASHAIR window during operation, an error message appears.
Subsequently, the FLASHAIR2 window is also closed.
Conversely, when I close the FLASHAIR2 window, an error message appears as well.
And the FLASHAIR window is closed in succession.

I think it is right that 
not only flashair window and flashair2 window simultaneously,
but also flashair window alone or flashair2 window alone must works well.

https://youtu.be/pEjkw-ic5I8
Is it a bug?

emoacht, 2020-08-19:
The situation you described is by itself normal. The multiple windows are shown by one instance of this app (only one instance can run at a time) and as a result, if an unmanageable error happens, the instance will be closed along with all windows.
in general, I don't recommend to close this app when it is doing some operations. Stopping operations gracefully may take some time. It is especially true in the case where network connection is involved.
Having said that, I am interested in the error message this app left. In such case, a variety of message may be thrown. If you still have it, could you post it?

emoacht, 2020-09-01:
@dentphoto Thank you for the log.
I made some changes to allow this app to close as gracefully as possible. I fixed some other bugs including exceptions at the timeout, unnecessary shortcut creation and activation after toast.
The store version is now in the pipeline for release.

closed 2020-09-07 by emoacht