emoacht/SnowyImageCopy #49: Report for option "Not select once-copied images on check"

dentphoto, 2020-09-03:
Snowy Plus 2.2.0 option "Not select once-copied images on check"
It doesn't work exactly.
However in the previous version  Snowy 1.9.0, it worked very accurately.

For verification, I did the following experiments.
1. Take 6 photos and wireless transferred.
2. Delete the photos sent to the local folder.
3. Close the Snowy app and Restart it.
4. Snowy 1.9.0 displays "once copied" and works correctly.
    However, Snowy PLUS randomly sends several files AGAIN. (Maybe a bug??)

# Snowy test1 : http://www.youtube.com/watch?v=bc-TyWxf2rc
# Snowy test2 : http://www.youtube.com/watch?v=XhKYhomT2bM
# Snowy Plus test1 : http://www.youtube.com/watch?v=2ECNlG9kycE
# Snowy Plus test2 : http://www.youtube.com/watch?v=w4qtwkN_3gA

dentphoto, 2020-09-04:
Even with Snowy Plus, it works normally if only one window is opened.
It seems to malfunction only when two or more windows are open.

emoacht, 2020-09-07:
@dentphoto Thank you for reporting.
It is a big. Working on.

emoacht, 2020-09-07:
I hope it is fixed by https://github.com/emoacht/SnowyImageCopy/commit/b6ae9ffcddd9e769bab24bc9bd6a6341c9584b42
I pushed this updated version into the pipeline.

dentphoto, 2020-09-07:
Thank you so much.
However, it has not been updated yet in Microsoft Store.

emoacht, 2020-09-09:
It seems to be published in the Store. 3 days. It took 3 days. so-so.

dentphoto, 2020-09-09:
It seems to work perfectly~^.^
Thank you.

# Snowy 2.2.1 => http://www.youtube.com/watch?v=3ebicntM14Q&t=15s

emoacht, 2020-09-10:
@dentphoto Thank you for confirming it.

closed 2020-09-10 by emoacht