emoacht/SnowyImageCopy #51: Request

dentphoto, 2020-09-09:
Hi~~Emoacht
What do you think about adding a few buttons to the switch box?
Present vs Wish
[image]
[image]

emoacht, 2020-09-10:
@dentphoto I had thought about the same thing when designing this menu but rejected it because closing a window should be done by the window's close button to prompt a user to make sure which window the user is about to close and whether the window is ready to be closed.

dentphoto, 2020-09-12:
Oh~You have considered it in advance.
Yes. Okay.

closed 2020-09-14 by emoacht