emoacht/SnowyImageCopy #52: Please let me know.

dentphoto, 2020-09-12:
What does "Automatic timeout ON/OFF" mean in AP mode setting?
And the unit is seconds? minute?

[image]

emoacht, 2020-09-14:
It's automatic timeout length in seconds.
Please refer to the help of [FlashAirTool](https://jp.toshiba-memory.com/ww/support/download/flashair/w04/config02.htm) by KIOXIA 

> （Auto Start Timeout Setting）
If there is no attempt to connect to the FlashAir™ card from a device (e.g. smartphone) within the time set in the timeout setting, the wireless LAN functionality automatically turns off.
To restart the wireless LAN functionality, turn off the device that the FlashAir™ is inserted into and then turn it on again.

dentphoto, 2020-09-14:
Thanks for kind answer.

closed 2020-09-14 by emoacht