emoacht/SnowyImageCopy #53: Snowy crash during operation in AP mode

dentphoto, 2020-09-14:
Now I'm sure Snowy Image Copy is the most powerful app for flashair.
I always thank for your energetic endeavor.
And I hope this is the last bug I found.^^

I am using flashair w-04 in "AP mode" and the wireless transmission to tablet PC is doing fine.
But, when the camera goes into Sleep mode (e.g. 1 minute) or
when I manually Turn off the camera during operation,
"Not Always but Occasionally" snowy windows close with an error message.
Please check this situation.

PS) No such error in Station mode

# related log files
[Sleep1-exception.log](https://github.com/emoacht/SnowyImageCopy/files/5218575/Sleep1-exception.log)
[Sleep2-exception.log](https://github.com/emoacht/SnowyImageCopy/files/5218576/Sleep2-exception.log)
[Sleep3-exception.log](https://github.com/emoacht/SnowyImageCopy/files/5218578/Sleep3-exception.log)
[Turn off1-exception.log](https://github.com/emoacht/SnowyImageCopy/files/5218579/Turn.off1-exception.log)
[Turn off2-exception.log](https://github.com/emoacht/SnowyImageCopy/files/5218580/Turn.off2-exception.log)
[Turn off3-exception.log](https://github.com/emoacht/SnowyImageCopy/files/5218581/Turn.off3-exception.log)

[image]

emoacht, 2020-09-24:
@dentphoto Thank you for reporting.
Although I could not reproduce it, I slightly modified exception handling.
https://github.com/emoacht/SnowyImageCopy/commit/148b77fa202acddf62e6e3a7d5216ad930e82ee5

dentphoto, 2020-09-24:
Unfortunately you cannot reproduce.
For reference I upload a video.
When the camera is turned off while operating in AP mode,
This is an error that occurs.

# https://youtu.be/J0NXSIPLkXQ

emoacht, 2020-09-24:
Thank you for the video.
Although it is hard to identify where the exception happens from the video, I could guess it from the log.
The modified version will be up in the Store in a few days as always.

emoacht, 2020-09-25:
Ver 2.2.2 become available in the Store. This time, it took less than half a day.

dentphoto, 2020-09-25:
Oh, very serious bug...
After running version 2.2.2, if you press the check button, copy button or check&copy button, Snowy dies due to a serious bug.
It dies after 30 seconds by pressing the auto button.

Couldn't you go back to Old 2.2.1 until the new version stabilizes?

emoacht, 2020-09-25:
Oh, it must be a lack of files in packages to the Store.
Working on.

emoacht, 2020-09-25:
Ver 2.2.3 is now in the Store. I hope it would work.

dentphoto, 2020-09-26:
Thanks for the quick action.
However, in sleep mode or in manual power off, there are still occasional errors.
Fortunately, the frequency of occurrence has decreased considerably.

emoacht, 2020-09-26:
Thank you for testing.
If the remaining errors are the same, it should not happen. Could you upload new logs again?

dentphoto, 2020-09-26:
AP mode (sleep mode/power off) bug that occurred very occasionally in Ver 2.2.3

[turn off-exception.log](https://github.com/emoacht/SnowyImageCopy/files/5286742/turn.off-exception.log)
[sleep-exception.log](https://github.com/emoacht/SnowyImageCopy/files/5286743/sleep-exception.log)
[sleep2-exception.log](https://github.com/emoacht/SnowyImageCopy/files/5286744/sleep2-exception.log)

(PS) Station mode is nearly perfect for all functions!!

emoacht, 2020-09-27:
Thank you for the logs.
Working on.

dentphoto, 2020-09-28:
Thanks~~Very good job !!
Finally, a minor bug in the flashair AP mode has been completely resolved.
Now Snowy Plus 2.2.4 is an omnipotent program.

emoacht, 2020-09-28:
Thank you for confirming.

closed 2020-09-28 by emoacht