emoacht/SnowyImageCopy #56: [Question] Where are the setting files saved?

dentphoto, 2020-09-30:
Before Snowy ver 1.9,  the setting files are saved in the following folder.
  [system drive]\Users\[user name]\AppData\Roaming\SnowyImageCopy\

Please let me know where is the setting files in Snowy Plus 2.2.4?

emoacht, 2020-09-30:
In general, the executable files and created files of a Store app are not supposed to be accessed by an user. They are handled by the OS and even a developer does not necessarily need to know their actual location.

Said that, the most part of Store apps' files are located in the following folder:
`C:\Users\[user name]\AppData\Local\Packages`
I am not sure but if you dig into the folders, you may be able to find it.

dentphoto, 2020-10-01:
Thank you...I've found it.
C:\Users\[user name]\AppData\Local\Packages\10186emoacht.SnowyImageCopy_0q7myvhtpbc7w\LocalCache\Roaming\SnowyImageCopy

closed 2020-10-02 by emoacht