emoacht/SnowyImageCopy #57: About flipped/rotated files

dentphoto, 2020-10-17:
Since the last bug fix, Snowy Plus 2.2.4 is always perfect without any errors.
Thanks for this.

Finally~~ I would like to ask you one question.

I have sent 5 photos. Then execute the actions.

First picture: Delete
Second picture: horizontal flip
Third picture: vertical flip
Fourth picture: clockwise rotation
Fifth picture: counterclockwise rotation

http://youtu.be/E-JfR7dnuoY

All five will be marked "Once Copied" in the Snowy next shooting session.

I understand intuitively the "Once Copied" mark on deleted photo thumbnails.

One curious thing...
Photos that are rotated or flipped will also change from "Copied" to "Once Copied".
Is it normal and reasonable to treat the rotated or flipped files like this?

dentphoto, 2020-10-18:
Oh ~ I knew it late.
Flip or rotate transforms the original image, so the file size also changes.
The only thing that remains after the original image is transformed, is the file with the same file name & different file size..
So Snowy recognizes that the original image was virtually gone
It is natural to mark it as “Once Copied”.
Is my understanding correct?

emoacht, 2020-10-19:
Yes, I guess so.
In theory, the data size of orientation of metadata is the same regardless of the value, a format change could happen when the rotation is edited.

closed 2020-10-20 by emoacht