emoacht/SnowyImageCopy #58: In Auto mode, "Once Copied" marking in real time

dentphoto, 2020-11-28:
emoacht, how are you?
I ask a very trivial and detailed question.

In Check&Copy mode...
When I delete the files already transferred and click Check&Copy,
Once Copied indication is displayed in real time.

But in Auto mode...
When I delete the files already transferred, Once Copied indication does not appear in real time.
"Once Copied" appears only when I manually press the Check (or Check&Copy) button after Stopping the Auto menu.

Even in Auto mode, can Once Copied marking be displayed in real time after deleting the transferred files? 
Without pressing the Check&Copy(or Check) button...

See the video below
https://youtu.be/KkbE6V4O2tU

emoacht, 2020-11-29:
I have been good. Thank you.
As for your question, it is probably because there is no new photo taken by the camera during the period. In auto mode, Snowy checks whether any change in the FlashAir and if no, it will return to waiting time. It is a key function to suppress unnecessary access to FlashAir in auto mode.

dentphoto, 2020-11-29:
Thank you for answer.

It is to prevent unnecessarily excessive access to card in auto mode.
Only when new files are found on the card, then Auto mode starts checking card and copy new files.

Did I understand well?

dentphoto, 2020-11-29:
I know Toshiba discontinued FlashAir in early 2019.
https://www.mouser.com/PCN/Toshiba_Memory_America_Inc_EOL_Notification_FlashAir_Jan_15_2019.pdf

So I don't have to wait for the W-05 generation anymore.

Now I have a powerful app like Snowy Plus, but I can't get FlashAir anymore.

So too sad.

71m383nd3r, 2020-11-29:
Still there are updates 
https://personal.kioxia.com/en-emea/software/flashair-configuration.html
https://personal.kioxia.com/en-emea/news.1.html

emoacht, 2020-11-29:
@dentphoto 

> Only when new files are found on the card, then Auto mode starts checking card and copy new files.

Your understanding is correct except checking will be triggered only when any write action has been done in FlashAir card.

I copied FlashAir product availability matter to another issue.

closed 2020-12-03 by emoacht