emoacht/SnowyImageCopy #60: Modification suggestion

vicentegr, 2020-11-29:
I use TP-LINK MR3020 as a WIFI repeater and I cannot access the card by name, the application does not let me put the ip of the card to locate it. It would be nice to be able to put the ip.
Sugerencia de modificación
Uso TP-LINK MR3020 como repetidor WIFI y no puedo acceder a la tarjeta por su nombre, la aplicación no me deja poner la ip de la tarjeta para localizarla. Estaría bien poder poner la ip.

emoacht, 2020-11-29:
Hello, @vicentegr 
How long the length of your IP address string is? Do you specify the port number?
You should be able to set a string up to 15, the maximum of ordinary IPv4 address, as the root of name field.

vicentegr, 2020-11-30:
Thank you very much for your attention.
The name field is auto-corrected to "http: // flashair /" automatically, to which I can add the folder structure I want. It is impossible to put an ip address.

vicentegr, 2020-11-30:
The WIFI distance reached by the flashair card inside the camera is minuscule, I place the portable router powered by a lithium battery next to the camera, so the signal reaches the equipment without problems. Download the application because it seems comfortable, but I ran into this problem.
I'm sorry for my terrible English, I use a translator.

emoacht, 2020-11-30:
In such case, I would suggest to write the whole path in the notepad and copy & paste it into the name textbox.

vicentegr, 2020-11-30:
Ocurrs the same

vicentegr, 2020-11-30:
An analysis of the entered text is programmed, which requires a certain format.

emoacht, 2020-11-30:
If you don't mind, could you write the path here?

vicentegr, 2020-11-30:
192.168.0.1

vicentegr, 2020-11-30:
It works, but it has to be written as http://192.168.0.1/

emoacht, 2020-11-30:
Okay. it is the usual way to write down URL path.

dentphoto, 2020-11-30:
Hi, emoacht~~

I found the difference in putting the ip address.
Are you intended to make this difference?

Snowy 1.9 - : 
http://192.168.0.140/dcim
                        http://flashair1/dcim  
                        http://flashair/ 
                     
Snowy 2.0+ : 
http://192.168.0.140/
                        http://flashair1/
                        http://flashair/ 

Only http://flashair/ can be used in both versions.

vicentegr, 2020-11-30:
Congratulations on the application, the operation is correct. Years ago I did not put the "http: //" when writing an ip. I suggest that when the program does not see the "http: //" add it, but do not add "flashair", at least as long as it is an ip. Someone else will the same thing happen to me. And thank you very much for your attention.

vicentegr, 2020-11-30:
"http://192.168.0.1/dcim" used in "Snowy + 2.2.4.0"

emoacht, 2020-11-30:
There should be no such difference.  Ver 2.* can accept the path after the root as well.
I have no plan to introduce a special treatment for IP address because I think it is relatively rare case and checking whether a path is IP address or not would require a bunch of code.

dentphoto, 2020-11-30:
That's weird.

In Snowy 1.9 and earlier, the address below does not work at all.
http://192.168.0.140/
http://flashair1/
I must add /dcim for flashair recognition.

On the other hand, in Snowy Plus, it works and transfers without /dcim.

dentphoto, 2020-11-30:
Hi? vicentegr~~

Attaching small wifi extender next to the camera for increasing the wireless range of the flashair card.
Same as my attempt.
https://cafe.naver.com/diffusercamfi/49

vicentegr, 2020-11-30:
Funny, I have a dental laboratory, hahaha.
The camera I use is the FUJIFILM S5pro with AF MICRO nikkor 60mm lens, it is the most suitable for my use, but I found the same problem, the Wi-Fi reach little, the router I use is very cheap and the change is spectacular. Now we only need to be able to delete the images of the flashair card from the computer, but I think this will no longer be achieved.

vicentegr, 2020-11-30:
To put the "flashair" on the S5pro you need a CF to SD adapter.
Extreme Compact Flash Type II Memory Card Adapter (Compatible with SDHC MMC SDHC SDXC) for Canon EOS 40d, 50d, 5d, 5D2, 350d, 400d; Sony A300, A700, A850, A900; Nikon D300, D70 Camera
This one worked

vicentegr, 2020-11-30:
Being inside the adapter, and the camera has a magnesium body, the signal is very small

dentphoto, 2020-11-30:
> Being inside the adapter, and the camera has a magnesium body, the signal is very small

After attaching the wifi extender next to the camera,  Did you configure the flashair card to "Station" mode?
Please show me your wifi extender+battery.

vicentegr, 2020-11-30:
[image]
The card is configured a few years ago, do not change the configuration to put the repeater.

vicentegr, 2020-11-30:
The battery also feeds a led ring, I do not need speed in the photo, the photo is static.

emoacht, 2020-11-30:
@vicentegr, @dentphoto, please don't mix up topics.
I am moving some comments to #61.

emoacht, 2020-11-30:
@dentphoto There should be no such difference. The path after the root address was implemented 5 years back by https://github.com/emoacht/SnowyImageCopy/commit/0add949f5c4d6a0412c4d7ea1d348766ec0b3edc

closed 2020-12-03 by emoacht