emoacht/SnowyImageCopy #67: Endless Memory...

dentphoto, 2021-09-03:
Hi, emoacht~~
Long time no see.

Endless memory function available in the past eye-fi sd card.
I think it would be great if it is possible in flashair w-04 as well.
Is it technically possible? 

[image]

emoacht, 2021-09-15:
Thanks for the suggestion.
I think there is no 100 percent reliable way to ensure that image files have been copied to local folder. So, yes, it is possible to delete older image files in FlashAir card regardless of whether or not they have been already copied.

dentphoto, 2021-09-15:
OK~~
Even if we can't be sure that the images have been copied with 100% certainty,
I think it will help the function to delete very old image files.
Please add the feature to the next version. 
Thanks.