emoacht/SnowyImageCopy #75: Migration to .NET 6.0

emoacht, 2022-01-20:
To assure longer application life cycle, I decided to migate this app to .NET 6.0, the latest platform which will be supported until 2024.
The downsides are:

- .NET 6.0 runtime is required.
- Package footprint increases dramatically.

dentphoto, 2022-01-23:
After running Snowy plus 3.0.0, press the Auto checking button.
Snowy crashes with an error message after 32 seconds .
Snowy normal 3.0.0 is the same.

Please confirm.

emoacht, 2022-01-23:
Thanks for the report as always.
Could you share the exception information?

dentphoto, 2022-01-23:
Here is log file.
[exception.log](https://github.com/emoacht/SnowyImageCopy/files/7919749/exception.log)
.

emoacht, 2022-01-23:
Thanks for the log.
It seems that the exception when the connection is unable has been changed.
Fixed by c41a12a8ac6fc049cb839967d38f2c5810da7249

emoacht, 2022-01-30:
It is observed that accessing FlashAir on .NET 6.0 suffers noticeable slow responces (e.g. around 1 second per access). See #77
I could not find a similar issue about HttpClient on .NET 6.0 and it is unlikely solved easily. Therefore, I decided to release both .NET 6.0 and .NET Framework 4.8 versions for the time being.

dentphoto, 2022-01-30:
[card scan speed]
Snowy plus3.1.1 Net48 = Snowy 3.1.0 Net48 >> Snowy 3.1.0 Net60
.Net60 and Snowy compatibility is definitely a problem.

dentphoto, 2022-01-30:
Standard 3.1.0 is ok, but plus 3.1.1 sometimes gives errors while scanning.

[plus 3.1.1 exception.log](https://github.com/emoacht/SnowyImageCopy/files/7965944/plus.3.1.1.exception.log)
 
[image]

emoacht, 2022-01-31:
Thanks for the report.
Unfortunately I am having a trouble with uploading updated package to Store.

emoacht, 2022-01-31:
Released minor update version.

dentphoto, 2022-02-01:
Thanks for the quick action.
Snowy plus 3.2.0Net48 & Snowy 3.2.0Net48
Both work very well.