emoacht/SnowyImageCopy #77: Snowy plus 3.0.1 is slower than Snowy 1.9.0

dentphoto, 2022-01-24:
Under the environment of same camera / same card setting / same network, 
Flashair card scan speed and file copy speed of Snowy plus 3.0.1...
It is considerably slower than that of Snow 1.9.0.

Why is this? 

# I will send you a movie if necessary.

emoacht, 2022-01-24:
I see. The code base of 3.x is almost identical with 2.x. 1.x is too old.
.NET 6.0 might affect the performance. Could you try the attached? It is a test version of .NET Framework 4.8.
[Snowy301.Net48.zip](https://github.com/emoacht/SnowyImageCopy/files/7925953/Snowy301.Net48.zip)

dentphoto, 2022-01-24:
Thanks for the quick response.
Scan speed and copy speed have been restored with .Net48.^^

[Performance difference]
Snowy301.Net48 = Snowy190 > Snowy plus 3.0.1

emoacht, 2022-01-24:
Thanks for the testing.
Then, enabling "Handle video file" makes any difference?

dentphoto, 2022-01-25:
Option [Handle video file] ON or OFF seems not to affect the performance of card scan & copy.

emoacht, 2022-01-28:
Thanks for the confirmation.
One more thing. Could you show the comparison between the version of .NET 6.0 (3.0.1) and that of .NET Framework 4.8 (3.0.2, as attached and has identical functionality with that of .NET 6.0) by concrete number (e.g. time length to copy the same files from FlashAir) with information on FlashAir version and your PC's spec and OS version?
[Snowy302.Net48.zip](https://github.com/emoacht/SnowyImageCopy/files/7956002/Snowy302.Net48.zip)

dentphoto, 2022-01-28:
**[My Flashair Card]**
Flashair W-04 16G
Fw 4.00.04 

**[PC spec]**
Edition : Windows 10 Home
Ver : 21H1
OS build : 19043.1466

**[Snowy 3.0.2 Net48  vs Snowy 3.0.1 Net60]**
https://youtu.be/_qoh-JF25XQ
3.0.1 Net60 is slow in checking and thumbnailing existing files when the camera is turned on.
Likewise, the recognition of real-time captured files is very slow, and as a result, the process feels slow.
Once recognized, there does not seem to be much difference in the copy speed itself between the two.
Crucially, it's clear that Net60 is noticeably slower in card scanning and thumbnailing...

emoacht, 2022-01-28:
Thanks for quick response.
To identify which part of the sequrence to access FlashAir takes longer time, could you enable `/recordlog` option and post the record so that I can compare with?

dentphoto, 2022-01-28:
[Net60 download.log](https://github.com/emoacht/SnowyImageCopy/files/7956857/Net60.download.log)

[Net48 download.log](https://github.com/emoacht/SnowyImageCopy/files/7956858/Net48.download.log)

emoacht, 2022-01-28:
Thanks.
Could you provide your PC's hardward spec (e.g. CPU, memory)?

dentphoto, 2022-01-28:
[image]

emoacht, 2022-01-28:
Thanks. Your PC seems to be powerful enough.
I found that on .NET 6.0, each step of accessing FlashAir itself is slow. I have never heard about it and there would be no clue.
The another differenece which I can think of is the platform (x86 or x64). So, could you try the version of .NET 5.0 for x86 (3.0.3, as attached) and the post the record as well?

dentphoto, 2022-01-28:
3.0.3 x86 has similar card scanning and thumbnailing performance to 3.0.1Net60.
Still, 3.0.2Net48 performs much better. 
 https://youtu.be/cWgPamTrRFY

[3.0.3 x86 download.log](https://github.com/emoacht/SnowyImageCopy/files/7959849/3.0.3.x86.download.log)

emoacht, 2022-01-29:
Thanks for the log.
I have no clue on this slow response at this moment.

emoacht, 2022-01-30:
As I could not figure out the root cause of this slow response, I decided to release both .NET 6.0 and .NET Framework 4.8 versions for minor update.
This issue is now closed. To continue the discussion, please post in #75

closed 2022-01-30 by emoacht