emoacht/SnowyImageCopy #81: My 2 suggestions

dentphoto, 2022-02-12:
[Suggestion 1]
The Standard version includes the first 7 basic options out of total 11 of Plus version.
I think it's appropriate to swap the positions of the following two options:
Because [Do not overwrite existing file] is an advanced user option that is sometimes necessary 
but infrequently preferred, it is proper to be included in the Plus version.
On the other hand, [Do not select once copied file on check] is the universal option to check most of the time,
so I think it's appropriate to include it in the Standard version.
What do you think about repositioning these two options? 

[Suggestion 2]
In [Do not overwrite existing file], "existing file" does not mean exactly the same file. 
However, some people misunderstand that it means the same file. 
How about replacing it with a detailed description? 
For example, 
[Do not overwite existing file of same filename but different exif] 

[image]

emoacht, 2022-02-13:
Thanks for the suggestions.
Suggestion 1: I don't think so.
Suggestion 2: I understand your point but it should be solved in other way.

emoacht, 2022-02-23:
Regarding suggestion 2, I guess that the value of date taken in Exif matches the time when a camera saves the image into FlashAir card. Of course, it depends on the camera model but checking my Canon's, they seem to match.
So, assuming that's the case, we can make use of the time to determine if the image is identical even if some changes, including the orientation, are added to the file.

dentphoto, 2022-02-23:
Without mentioning the exif, 
Can we simply think of [Do not overwrite existing file of same filename but different file size]?

emoacht, 2022-02-23:
It looks confusing and will raise question about the case of the same file name and same file size.

emoacht, 2022-02-24:
I added explanation to readme. I hope a user to refer it when the user has any question on this.

closed 2022-02-24 by emoacht