emoacht/SnowyImageCopy #83: "Once Copied" marking on thumbnail

dentphoto, 2022-02-23:
When files that have already been transferred to local folder are deleted or flipped or rotated,
It's a great feature that snowy's existing thumbnails display  "Once Copied" in the next transfer session.

In Snowy 1.9 or 3.2.0 it has been perfect. 
https://youtu.be/_i1x63sRQZI
https://youtu.be/YpEQ8cQoqCQ

But, In Snowy 3.3.0, this "Once Copied" marking on thumbnails is unstable.  Randomly Some "Once Copied"...Others "Copied"... 
https://youtu.be/_6mKu4S1cX0
https://youtu.be/B1gij68A5Q8
https://youtu.be/xgd09tSG6-s

Please confirm.

dentphoto, 2022-02-24:
Thanks for the quick action.
Snowy plus 3.3.1 works perfect as previous versions.

emoacht, 2022-02-24:
Thanks for the confirmation.

closed 2022-02-24 by emoacht