emoacht/SnowyImageCopy #89: Multi Memory

termitaklk, 2022-12-14:
Good afternoon,

The program works great! Is there a possibility that it supports multiple memories and not just one?

termitaklk, 2022-12-14:
Hello,

I mean multiple sd cards flashair. Right now I can only do it one by one, is it possible to add more than 1 memory card at a time to copy?

termitaklk, 2022-12-14:
Or is it possible that the app can be opened several times at the same time?

emoacht, 2022-12-15:
I see. Simultaneous access to multiple FlashAir cards is the feature of Microsoft Store version. It is paid though.

termitaklk, 2022-12-15:
Yes, I just saw it and I'm about to buy it. I would like to give you some suggestions to make the app more comfortable for companies that use these SD cards.

1. That the execution of multiple memories be in a single window, because with 2 or 3 it is not a problem but with 16, for example, there are 16 open windows and it is chaotic.

2. It would be great if the app works as a windows service too or has the option to start with windows.

3. If it is possible to integrate mail or windows notifications when a memory sd is offline.

emoacht, 2022-12-15:
I have no plan for major update but thanks for the suggestions anyway.

1. Wow. I have never imagined that accessing 16 cards simultaneously.

2. I don't see a particular reason to run as a service. You can start this app from command-line with `/minimized` option not to show the window. This is explained in readme.

3. I remeber that I decided not to add such function because it is hard to detect if a particular card is online or not when the card is not directly connected to the system.

termitaklk, 2022-12-16:
Thanks for answering, I'll tell you. Flashairs are a very effective solution for excursion photography sales projects. Tourism increases every year more and companies seek faster and more effective solutions. Currently there are systems that integrate with flashair memories and upload and classify everything automatically, but they are projects of hundreds of thousands of dollars. Your software has everything that is needed for this purpose but with certain adjustments and improvements. Here where I am we have 20 photographers using flashair, but there are other companies with the same or greater number of photographers.

Seeing it from this angle, it is clear that it needs to work as a service, because it is installed on a server, the SD cards are linked and the routes to which to send the photos are configured and if any failure is reported to solve the problem. I mean, you see the software in a way that a user would be manipulating it, I see it as an independent solution that should only be monitored from time to time.